DRCL-USC/Hector_Simulation #2: and there is an problem like that when i press the keyborad up and down key，the robot walk will fall down

dbdxnuliba, 2023-10-24:
thanks for your nice work
and there is an problem like that when i press the keyborad up and down key，the robot walk will fail
[image]

yiyuc, 2023-10-24:
Hi, in src/interface/keyboard.cpp, we defined w and s keys for forward and backward velocity commands. We are checking both from the command interface side and the controller side to fix it.

junhengl, 2023-10-25:
Up and down arrow keys shouldn't do anything. I have tested the other keyboard commands, it seems working fine for me. The only problem was the velocity command when using the keyboard was a bit large, please check out the new update. The keyboard control instruction is also updated in the readme file.

closed 2023-11-08 by junhengl