DRCL-USC/Hector_Simulation #7: error: RLException: [biped.launch] is neither a launch file in package [unitree_gazebo] nor is [unitree_gazebo] a launch file name The traceback for the exception was written to the log file

wei1224hf, 2023-11-05:
after I compile the code , then try to run it , 
`
roslaunch unitree_gazebo biped.launch
RLException: [biped.launch] is neither a launch file in package [unitree_gazebo] nor is [unitree_gazebo] a launch file name
The traceback for the exception was written to the log file

`

wei1224hf, 2023-11-05:
没有 unitree_gazebo  这个包啊， 奇怪了

roscd unitree_gazebo
roscd: No such package/stack 'unitree_gazebo'

wei1224hf, 2023-11-05:
也不知怎么整了一下， 可以跑起来了， 但是仍旧报错：

`
oslaunch unitree_gazebo biped.launch
... logging to /home/ubuntu/.ros/log/42d5c8a0-7bdb-11ee-830f-d720a2f6b59c/roslaunch-ml-ubuntu20-04-desktop-v1-0-54gb-100m-5553.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
substitution args not supported:  No module named 'defusedxml'
when processing file: /home/ubuntu/catkin_ws/src/hector_description/xacro/robot.xacro
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '--inorder', '/home/ubuntu/catkin_ws/src/hector_description/xacro/robot.xacro', 'DEBUG:=false']] returned with code [2]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro --inorder '$(arg dollar)$(arg robot_path)/xacro/robot.xacro'             DEBUG:=$(arg user_debug)"/>
The traceback for the exception was written to the log file
`

junhengl, 2023-11-05:
Are you using ros melodic?

wei1224hf, 2023-11-06:
> Are you using ros melodic?

no , using ros noetic

yiyuc, 2023-11-19:
Please configure your ros workspace properly, this looks like some conflicts within the unitree_gazebo package. Since the Hector robot is built based on the A1 robot, we adopted the environment from unitree_ros package. If you have other Unitree related packages in the same workspace or somewhere, this issue might happen.

closed 2023-11-30 by junhengl