DRCL-USC/Hector_Simulation #11: confusion about Constraints Matrix

BIT-XLJ, 2023-12-17:
Hello, thanks for your awesome job!
But I have some questions about the constraints matrix in SolverMPC.cpp。
[image]
I know the first four rows of F_control are friction pyramid constraints, and the fifth row of F_control aims to guarantee the moment of X axis equals to zero. But what about the function of the last three rows of F_control? I didn't find the details of this code clip in your essay. Could you please explain this to me? Or can somebody help me?

junhengl, 2023-12-17:
Hi, row 5&6 are part of line foot constraint for ensuring no toe/heel lifts. You can read more about this in [1]. The last row is force saturation constraint. We put 2 in the Fz slot to avoid bugs in the variable elimination process. 

[1] Caron, S., Pham, Q. C., & Nakamura, Y. (2015, May). Stability of surface contacts for humanoid robots: Closed-form formulae of the contact wrench cone for rectangular support areas. In 2015 IEEE International Conference on Robotics and Automation (ICRA) (pp. 5107-5112). IEEE.

BIT-XLJ, 2023-12-18:
Thank you some much!

BIT-XLJ, 2023-12-18:
Thank you so much!

closed 2023-12-18 by BIT-XLJ