DRCL-USC/Hector_Simulation #12: About the horizonLength of MPC controllor

gaiyi7788, 2023-12-21:
Outstanding work!

I hope to try changing the horizonLength of the MPC controllor to see how the controllor works. However, when I turn the horizonLength to less than 10, the following error happened:
`hector_ctrl: malloc.c:4036: _int_malloc: Assertion `(unsigned long) (size) >= (unsigned long) (nb)' failed.`

Could you give me some advice, thank you!

junhengl, 2023-12-21:
Hi there, this code structure is not really meant to be used with shorter than 10 horizon length. You may want to rewrite some of the gait planning and contact schedule section of the code to reach this autonomy. Feel free to modify those in convexMPCLocomotion.cpp

gaiyi7788, 2023-12-22:
Got it! Thank you for the timely reply!

closed 2023-12-22 by gaiyi7788