DRCL-USC/Hector_Simulation #13: About the MPC sampling frequency and the velocity tracking.

gaiyi7788, 2023-12-25:
I'm sorry to bother you, but I have the following two confusions hoping to be addressed.

The first question is about the MPC sampling frequency. In the code the variable _iterations_between_mpc is set to 40 i.e. sampling frequency 25hz, as opposed to 33.3 Hz.. Is it due to the stability of the MPC solution?

The second question is about the velocity tracking. When I set the velocity command to vx=0.5, the centre of mass velocity is probably only about 0.3, which doesn't quite match the velocity tracking results from the paper. Can you give me some suggestions to achieve better velocity tracking?

Again, sorry to bother you and look forward to hearing from you!

junhengl, 2023-12-25:
Hi, thanks for the questions. 
1. you could set mpc sampling frequency freely in this framework. For example, you can run mpc with 40ms sampling rate and use iteration_between_mpc any int number from 1 to 40. The lower the number, the better the mpc reacts to commands. 
2. The velocity tracking is somewhat dependent on the gain tuning on com position state tracking. Normally in this framework I don’t let it track velocity directly, but tracking the position trajectory from velocity commands. I set somewhat conservative and balanced weights in current mpc code, however, you may achieve much better velocity tracking with high com_x weight in mpc and fine tune the foot heuristic landing position. 

Hope it helps.

gaiyi7788, 2023-12-25:
Thank you very much for your reply!

closed 2023-12-25 by gaiyi7788