DRCL-USC/Hector_Simulation #14: Is the mechanical design available?

zitongbai, 2024-01-20:
Hello 

Thanks for the wonderful work!

I am interested in humanoid robot and have roughly skimmed your articles and repo, but I can only find the mesh you used in Gazebo simulation. I want to make my own robot in the real world and I wonder if the mechanical design (like solidworks files) is available.

Thanks for your attention.

junhengl, 2024-01-20:
Hello,
Hardware wise, Only major link parts in stl are available in this project. SW files are not open source as of now. You are welcomed to recreat and mimic hector parts on your own. However, HECTOR robots are also for sale if you are interested. https://laser-robotics.com/?page_id=362
Junheng

zitongbai, 2024-01-21:
Thanks for your prompt and informative response. Your guidance is invaluable as I delve into this project.

Best regards

closed 2024-01-21 by zitongbai