DRCL-USC/Hector_Simulation #19: Unable to make

goodmorning-hwt, 2024-04-10:
I was trying to build this repository on my computer(Ubuntu 20.04 ROS Neotic) but the output is like this

```bash
catkin_make -DCMAKE_BUILD_TYPE=Release
Base path: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws
Source space: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/src
Build space: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/build
Devel space: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel
Install space: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/install
####
#### Running command: "cmake /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DCATKIN_DEVEL_PREFIX=/home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/install -G Unix Makefiles" in "/home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/build"
####
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/hwt-ubuntu/Documents/limxdynamics/pointfoot-gazebo-ros/install;/opt/ros/noetic
-- This workspace overlays: /home/hwt-ubuntu/Documents/limxdynamics/pointfoot-gazebo-ros/install;/opt/ros/noetic
-- Found PythonInterp: /home/hwt-ubuntu/miniconda3/envs/ros-neotic/bin/python3 (found suitable version "3.8.18", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /home/hwt-ubuntu/miniconda3/envs/ros-neotic/bin/python3
-- Using Debian Python package layout
-- Found PY_em: /home/hwt-ubuntu/miniconda3/envs/ros-neotic/lib/python3.8/site-packages/em.py
-- Using empy: /home/hwt-ubuntu/miniconda3/envs/ros-neotic/lib/python3.8/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /home/hwt-ubuntu/miniconda3/envs/ros-neotic/bin/python3 (found version "3.8.18")
-- Found Threads: TRUE
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 6 packages in topological order:
-- ~~  - hector_description
-- ~~  - unitree_legged_msgs
-- ~~  - unitree_legged_control
-- ~~  - hector_control
-- ~~  - unitree_controller
-- ~~  - unitree_gazebo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'hector_description'
-- ==> add_subdirectory(hector_description)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'unitree_legged_msgs'
-- ==> add_subdirectory(unitree_ros/unitree_legged_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- unitree_legged_msgs: 11 messages, 0 services
-- +++ processing catkin package: 'unitree_legged_control'
-- ==> add_subdirectory(unitree_ros/unitree_legged_control)
-- +++ processing catkin package: 'hector_control'
-- ==> add_subdirectory(hector_control)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
[... 338 lines cut ...]
[ 55%] Generating Javascript code from unitree_legged_msgs/LowState.msg
[ 56%] Generating Lisp code from unitree_legged_msgs/HighState.msg
[ 57%] Generating Javascript code from unitree_legged_msgs/HighCmd.msg
[ 58%] Generating Lisp code from unitree_legged_msgs/HighCmd.msg
[ 60%] Generating EusLisp code from unitree_legged_msgs/HighCmd.msg
[ 59%] Generating EusLisp code from unitree_legged_msgs/LowState.msg
[ 61%] Generating EusLisp code from unitree_legged_msgs/HighState.msg
[ 61%] Generating EusLisp manifest code for unitree_legged_msgs
[ 62%] Generating Javascript code from unitree_legged_msgs/HighState.msg
[ 62%] Built target unitree_legged_msgs_generate_messages_lisp
[ 63%] Generating Python msg __init__.py for unitree_legged_msgs
[ 63%] Built target unitree_legged_msgs_generate_messages_nodejs
[ 63%] Built target unitree_legged_msgs_generate_messages_py
[ 63%] Built target unitree_legged_msgs_generate_messages_eus
[ 64%] Linking CXX executable /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel/lib/unitree_controller/unitree_move_kinetic
[ 64%] Built target unitree_move_kinetic
[ 65%] Linking CXX executable /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel/lib/unitree_controller/unitree_external_force
[ 65%] Built target unitree_external_force
[ 67%] Linking CXX shared library /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel/lib/libqpOASES.so
[ 67%] Built target qpOASES
[ 68%] Linking CXX shared library /home/hwt-ubuntu/Documents/limxdynamics/reference/Hector/catkin_ws/devel/lib/libunitreeFootContactPlugin.so
[ 68%] Built target unitreeFootContactPlugin
make: *** [Makefile:141: all] Error 2
Invoking "make -j24 -l24" failed
```

junhengl, 2024-04-14:
have you tried building it within the \catkin_ws instead of \catkin_ws\build?

xiaohai-AI, 2024-05-01:
AttributeError: module 'em' has no attribute 'RAW_OPT'
reinstall ros-noetic may be ?