DRCL-USC/Hector_Simulation #21: catkin_make error

abnerjie, 2024-05-14:
Hello, I am encountering some issues while compiling my project. When I run catkin_make, I run into a problem. The same problem occurs when I compile with catkin_make -DCMAKE_BUILD_TYPE=Release. The issue is that it prompts with the following message:
"In file included from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/interface/../../include/interface/../sdk/include/unitree_legged_sdk/unitree_legged_sdk.h:12,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/interface/../../include/interface/CmdPanel.h:38,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/interface/../../include/interface/KeyBoard.h:11,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/interface/KeyBoard.cpp:1:
/home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/interface/../../include/interface/../sdk/include/unitree_legged_sdk/lcm.h:9:10: fatal error: lcm/lcm-cpp.hpp: 没有那个文件或目录
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/build.make:310：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/src/interface/KeyBoard.cpp.o] 错误 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/../sdk/include/unitree_legged_sdk/unitree_legged_sdk.h:12,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/CmdPanel.h:38,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/DesiredCommand.h:15,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/ControlFSMData.h:4,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/FSMState.h:6,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/FSM.h:4,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/FSM.cpp:1:
/home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/../sdk/include/unitree_legged_sdk/lcm.h:9:10: fatal error: lcm/lcm-cpp.hpp: 没有那个文件或目录
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
In file included from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/../sdk/include/unitree_legged_sdk/unitree_legged_sdk.h:12,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/CmdPanel.h:38,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/DesiredCommand.h:15,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/ControlFSMData.h:4,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/FSMState.h:6,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/FSMState.cpp:1:
/home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/../sdk/include/unitree_legged_sdk/lcm.h:9:10: fatal error: lcm/lcm-cpp.hpp: 没有那个文件或目录
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/build.make:154：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/src/FSM/FSM.cpp.o] 错误 1
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/build.make:167：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/src/FSM/FSMState.cpp.o] 错误 1
In file included from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/../include/common/../interface/../sdk/include/unitree_legged_sdk/unitree_legged_sdk.h:12,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/../include/common/../interface/CmdPanel.h:38,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/../include/common/DesiredCommand.h:15,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/../include/common/ControlFSMData.h:4,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/main.cpp:7:
/home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/../include/common/../interface/../sdk/include/unitree_legged_sdk/lcm.h:9:10: fatal error: lcm/lcm-cpp.hpp: 没有那个文件或目录
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/build.make:63：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/src/main.cpp.o] 错误 1
In file included from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/common/../../include/common/../interface/../sdk/include/unitree_legged_sdk/unitree_legged_sdk.h:12,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/common/../../include/common/../interface/CmdPanel.h:38,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/common/../../include/common/DesiredCommand.h:15,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/common/DesiredCommand.cpp:1:
/home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/common/../../include/common/../interface/../sdk/include/unitree_legged_sdk/lcm.h:9:10: fatal error: lcm/lcm-cpp.hpp: 没有那个文件或目录
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/build.make:219：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/src/common/DesiredCommand.cpp.o] 错误 1
In file included from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/../sdk/include/unitree_legged_sdk/unitree_legged_sdk.h:12,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/CmdPanel.h:38,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/DesiredCommand.h:15,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/ControlFSMData.h:4,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/FSMState.h:6,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/FSMState_Passive.h:4,
                 from /home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/FSMState_Passive.cpp:1:
/home/xie/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/hector_control/src/FSM/../../include/FSM/../common/../interface/../sdk/include/unitree_legged_sdk/lcm.h:9:10: fatal error: lcm/lcm-cpp.hpp: 没有那个文件或目录
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
[... cut ...]
    9 | #include <lcm/lcm-cpp.hpp>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/build.make:297：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/src/interface/CheatIO.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:9780：Hector_Simulation/Hector_ROS_Simulation/hector_control/CMakeFiles/hector_ctrl.dir/all] 错误 2
make: *** [Makefile:141：all] 错误 2
Invoking "make - -j20 -l20" failed
"

yiyuc, 2024-05-15:
install lcm from here https://lcm-proj.github.io/lcm/

OSKOOO, 2024-06-27:
I will close this, feel free to let us know if you run into any trouble.

closed 2024-06-27 by OSKOOO