DRCL-USC/Hector_Simulation #22: catkin build "make -j4 -l4 failed"

tien-li, 2024-05-29:
/usr/bin/ld: skipping incompatible /home/ken/Desktop/catkin_ws/src/Hector_Simulation/Hector_ROS_Simulation/unitree_ros/unitree_legged_control/lib/libunitree_joint_control_tool.so when searching for -lunitree_joint_control_tool
/usr/bin/ld: cannot find -lunitree_joint_control_tool
collect2: error: ld returned 1 exit status
make[2]: *** [Hector_Simulation/Hector_ROS_Simulation/unitree_ros/unitree_legged_control/CMakeFiles/unitree_legged_control.dir/build.make:110: /home/ken/Desktop/catkin_ws/devel/lib/libunitree_legged_control.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:2532: Hector_Simulation/Hector_ROS_Simulation/unitree_ros/unitree_legged_control/CMakeFiles/unitree_legged_control.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed

I try to catkin_build on my computer but I get this issue, do you know how to fix this?

tien-li, 2024-05-29:
I am using M2 Macbook pro with Ubuntu 20.04 on virtual machine

junhengl, 2024-05-30:
Unfortunately ARM platforms are not well-supported. You will run into many issues.

OSKOOO, 2024-06-27:
I will close this issue.

closed 2024-06-27 by OSKOOO