DRCL-USC/Hector_Simulation #24: question

tangwuliuliu, 2024-07-11:
I want to know which parts of the code I should modify to adapt the controller to my own model if I import my own robot urdf ？

kingnight797, 2024-07-15:
hector_control

yili5912, 2024-07-22:
Hello,
I just cloned Hector_Simulation and compiled OK. After Click the start button at the bottom of the simulator, the robot fall down. Which part should I change to make it walk?
Thanks!

junhengl, 2024-08-01:
You can first update robot model in hector_description/xacro. To make the controllers compatible with the new model, you will also need to modify the files in hector_control with new parameters from your robot.

tangwuliuliu, 2024-09-04:
> hector_control

Sorry to bother you again, I would like to ask how you do the slope test. When I use this method to do the slope test, the performance is always OK at the beginning, but it shakes violently as I continue to go up