DRCL-USC/Hector_Simulation #25: Robot fall down after clicking the start button

yili5912, 2024-07-23:
Hello,
I just cloned Hector_Simulation and compiled OK. After Click the start button at the bottom of the simulator, the robot fall down. Which part should I change to make it walk?
Thanks!

junhengl, 2024-07-28:
You can reset the controller and then reset the robot with ctrl+R after it fails.

yili5912, 2024-07-28:
I did it many times. Only one time it didn't fall down. Where can I check to make sure the controller is running correctly ?
Thanks.

junhengl, 2024-07-28:
What's your system specs? the simulation runs pretty CPU and GPU-intensive, could you try lower the real time factor and see if performance improves?

yili5912, 2024-07-28:
Here is my laptop info: 
model name      : Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz, 
MemTotal:        8015004 kB

What real time factor do you suggest ? And where is the code?
Thanks!

junhengl, 2024-07-28:
You can change real-time factor on the left panel of the gazebo GUI, or in the hector launch file. Your PC specs are not recommended to run this simulation in real-time. You can start with 1/10 of real-time and adjust based on its performance.

yili5912, 2024-07-29:
Thank you very much. I changed it to 0.3 and it works now. 
Regards.

yili5912, 2024-07-29:
The issue is solved. Thanks.

closed 2024-07-29 by yili5912