DRCL-USC/Hector_Simulation #26: how to understand "near_two" in this nice project

seek66, 2024-07-31:
想问下为啥用“near_two(c_row[j])”这个条件标记要消除的变量呢，为啥是数值接近2呢
[image]

junhengl, 2024-08-01:
Hi, 
The original Cheetah source code used near_one as the condition to do elimination, and that indicates the force limit constraint for each leg. However, on Hector implementation, we have additional constraints that may have 1s in constraints other than just force limits. for example,  https://github.com/DRCL-USC/Hector_Simulation/blob/5e0ca67ce4ca5b1879391438c6f54c7224ec4da1/Hector_ROS_Simulation/hector_control/ConvexMPC/SolverMPC.cpp#L524 
To work around it, we use near_two to check when the force limit constraint appears in order to do eliminations. In the F_control matrix, you can find the corresponding lines that have 2 in them, indicating the force limit constraint to do the elimination checking. 
https://github.com/DRCL-USC/Hector_Simulation/blob/5e0ca67ce4ca5b1879391438c6f54c7224ec4da1/Hector_ROS_Simulation/hector_control/ConvexMPC/SolverMPC.cpp#L528

seek66, 2024-08-02:
thank you very very very very much！！ 

黎健亮
***@***.***

 

 https://github.com/DRCL-USC/Hector_Simulation/blob/5e0ca67ce4ca5b1879391438c6f54c7224ec4da1/Hector_ROS_Simulation/hector_control/ConvexMPC/SolverMPC.cpp#L528

seek66, 2024-08-02:
> Hi, The original Cheetah source code used near_one as the condition to do elimination, and that indicates the force limit constraint for each leg. However, on Hector implementation, we have [...]

thank you  very very very much!!

closed 2024-08-06 by junhengl