DRCL-USC/Hector_Simulation #27: how to run in ros2

Lau-JW, 2024-09-05:
I want to show all my respect to you, and how to run your project in ros2 whose version  unitree_ros does't support

junhengl, 2024-09-12:
Thanks, we have not planned to migrate this simulation to ros2 yet. Please stay tuned for the updates. 

- Junheng

krewm271, 2024-10-15:
i have make similar project for 6 DOF leg bipedal robot in ros2 it is trying to balance but robot is continously falling. is there any idea how i can fix it (for the mpc controller i have used hector_control only)

junhengl, 2024-10-15:
@krewm271 6DoF legs will have a different foot kinematics constraint than 5-DoF leg + line foot. To start, you can try to delete the line foot constraint and the zero moment x constraint.  -Junheng